jkent/frogfs #8: How to get espfs image size?

mharizanov, 2019-09-28:
I need to know the espfs image size, so I can calculate MD5 on it and check if updating it over OTA was successful.
Is there a way to get the espfs image size other than traversing through the included files?

Cheers

jkent, 2019-09-28:
Yes, at the moment you can use `image_espfs_size` or `image_espfs_end - image_espfs_start` to get the size using `espfs_image.h`.  The interface will be changing in the future (wip branch) to `espfs_image_bin_len` because xxd is being used to include espfs binary.

Hope that helps!

jkent, 2019-09-28:
I just re-read your comment and I realized you want to OTA the espfs binary separate from the firmware binary?  Is that correct?  There currently isn't a provision for that, but it could be implemented.

mharizanov, 2019-09-28:
Yes, exactly. It is flashed on a separate partition and not embedded into the binary, so your original proposal won't work for this use case. 
OK, I'll just go for iterating and summing up the file sizes.
Thank you!

closed 2019-09-28 by mharizanov

jkent, 2019-09-28:
I'm going to create a new issue, so this gets first class support.  I never really thought about having espfs OTA'd separate from the application binary.