jkent/frogfs #14: Error on CMake build

Elektrik1, 2019-10-08:
This error is being generated on build with CMake (ESP-IDF Tools 2.1). It work fine thou when using legacy make with exact same sources.

```
[1059/1109] Performing configure step for 'mkespfsimage'
-- Configuring done
-- Generating done
-- Build files have been written to: D:/Arduino/Builds/concert_bt_cmake/esp-idf/esp32-espfs/mkespfsimage-prefix/src/mkespfsimage-build
[1062/1109] Performing build step for 'mkespfsimage'
[1/2] Building C object CMakeFiles/mkespfsimage.dir/main.c.obj
[2/2] Linking C executable mkespfsimage.exe
[1089/1109] Performing install step for 'mkespfsimage'
[1/3] Building C object CMakeFiles/mkespfsimage.dir/main.c.obj
[2/3] Linking C executable mkespfsimage.exe
[2/3] Install the project...
-- Install configuration: ""
-- Installing: D:/Arduino/Builds/concert_bt_cmake/esp-idf/esp32-espfs/bin/mkespfsimage.exe
[1098/1109] Generating espfs_image.bin
FAILED: esp-idf/esp32-espfs/espfs_image.bin
cmd.exe /C "cd /D D:\Arduino\Builds\concert_bt_cmake\esp-idf\esp32-espfs && D:\Espressif\.espressif\tools\cmake\3.13.4\bin\cmake.exe -E env BUILD_DIR=D:/Arduino/Builds/concert_bt_cmake/esp-idf/esp32-espfs CONFIG_ESPFS_PREPROCESS_FILES= CONFIG_ESPFS_CSS_MINIFY_UGLIFYCSS=n CONFIG_ESPFS_HTML_MINIFY_HTMLMINIFIER=n CONFIG_ESPFS_JS_CONVERT_BABEL=n CONFIG_ESPFS_JS_MINIFY_BABEL=n CONFIG_ESPFS_JS_MINIFY_UGLIFYJS=n CONFIG_ESPFS_UGLIFYCSS_PATH= CONFIG_ESPFS_HTMLMINIFIER_PATH= CONFIG_ESPFS_BABEL_PATH= CONFIG_ESPFS_UGLIFYJS_PATH= D:/Dropbox/Arduino/Sketches/Projects/ESP32/A2DP/a2dp_sink/components/esp32-espfs/tools/build-image.py D:/Dropbox/Arduino/Sketches/Projects/ESP32/A2DP/a2dp_sink/html"
%1 is not a valid Win32 application
[1099/1109] Linking CXX static library esp-idf\esp_https_ota\libesp_https_ota.a
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

jkent, 2019-10-13:
Hi @Elektrik1, I've got some time, I'll boot into Windows and see what I can do.

jkent, 2019-10-13:
Thanks for your bug report! :)

commit e9ebf266 (2019-10-13): shebangs don't work in Windows, use build property PYTHON, closes #14
closed 2019-10-13 by jkent