jkent/frogfs #21: build error

phatpaul, 2020-02-11:
Win 7.  Legacy GNU Make (msys2/mingw32).  ESP-IDF 3.3.1.

I unselected menuconfig->components->espfs->preprocess files.
Leads to:
```
#
# espfs
#
CONFIG_ESPFS_MAX_PARTITIONS=3
CONFIG_ESPFS_IMAGEROOTDIR="html"
CONFIG_ESPFS_PREPROCESS_FILES=
CONFIG_ESPFS_USE_HEATSHRINK=y
CONFIG_ESPFS_USE_GZIP=y
CONFIG_ESPFS_LINK_BINARY=y
```
get build error:
```
make[1]: *** No rule to make target 'node_modules', needed by 'espfs_image.bin'.  Stop.
```

Workaround with:
```
#
# espfs
#
CONFIG_ESPFS_MAX_PARTITIONS=3
CONFIG_ESPFS_IMAGEROOTDIR="html"
CONFIG_ESPFS_PREPROCESS_FILES=y
CONFIG_ESPFS_CSS_MINIFY_NONE=y
CONFIG_ESPFS_CSS_MINIFY_UGLIFYCSS=
CONFIG_ESPFS_HTML_MINIFY_NONE=y
CONFIG_ESPFS_HTML_MINIFY_HTMLMINIFIER=
CONFIG_ESPFS_JS_CONVERT_NONE=y
CONFIG_ESPFS_JS_CONVERT_BABEL=
CONFIG_ESPFS_JS_MINIFY_NONE=y
CONFIG_ESPFS_JS_MINIFY_BABEL=
CONFIG_ESPFS_JS_MINIFY_UGLIFYJS=
CONFIG_ESPFS_USE_HEATSHRINK=y
CONFIG_ESPFS_USE_GZIP=y
CONFIG_ESPFS_LINK_BINARY=y
```
Builds OK (CONFIG_ESPFS_PREPROCESS_FILES is y, but all sub-options off)

phatpaul, 2020-02-11:
forgot to mention testing commit: 76aca787824fcb0bec507a60117bb5858d3edf6a

Shaopus, 2020-02-24:
I set up the same as you ,but found this error:`/usr/bin/env: “python3”: No such file or directory`

closed 2020-03-31 by phatpaul