jkent/frogfs #28: Mmap possible ?

X-Ryl669, 2021-12-20:
As far as I understand, esp-idf support `esp_partition_mmap` function, and espfs is mostly already a memory mapped filesystem.
Is is possible to link both to get a partition on flash containing RO data we could access via mmap (that is getting access to `espfs_file_t->raw_start`  and `length`) ?

I see multiple advantages to this scheme:
1. No need to copy the data from ROM to RAM (since RAM is very limited on ESP32) like with the VFS layer. Once could use the raw pointer as if it was a C array
2. Data is update-able via OTA (provided the right code is written for writing to the right partition). Obviously, care must be taken not to read from the partition while it's updated.
3. All other advantages to use ESPFS.

jkent, 2022-02-20:
There is already a function for this.  It is espfs_faccess.  It's usage is slightly different from your mmap implementation.

```C
/**
 * \brief Get raw memory for an uncompressed open file object
 *
 * \return length of file or < 0 upon error
 */
ssize_t espfs_faccess(
    espfs_file_t *f, /* [in] espfs file */
    void **buf /** [out] doube pointer to buf */
);
```

closed 2022-02-20 by jkent

X-Ryl669, 2022-02-20:
How could I've missed that ?
Thanks!