jkent/frogfs #29: Chosen hash function is colliding

X-Ryl669, 2021-12-20:
Currently chosen hash function is colliding. In effect, any char X followed by Y so that Y = X+33 gives the same result as X+1 followed by X.
And since a path is very likely to have char only separated by 33, this is going to happen frequently. 

Replacing:
```cpp
static uint32_t hash_path(const char *path)
{
    uint32_t hash = 5381;
    const uint8_t *p = (uint8_t *)path;

    while (*p) {
        /* hash = hash * 33 + *p */
        hash = (hash << 5) + hash + *p;
        p++;
    }

    return hash;
}
```
by
```cpp
static uint32_t hash_path(const char *path)
{
    uint32_t hash = 5381;
    const uint8_t *p = (uint8_t *)path;

    while (*p) {
        /* hash = hash * 257 + *p */
        hash = (hash << 8) + hash + *p;
        p++;
    }

    return hash;
}
```
will solve this issue.

jkent, 2022-02-20:
Thank you!  I'm not sure where I pulled the original hashing function from, but this does make way more sense.  Would you mind updating your pull request and increment ESPFS_VERSION_MAJOR?

jkent, 2022-03-08:
Resolved, thank you again.

closed 2022-03-08 by jkent