jkent/frogfs #38: size of gzip files is incorrect

phatpaul, 2022-10-12:
I noticed my http server (https://github.com/chmorgan/libesphttpd) sending extra junk data after the contents of a gzip file.  In fact, it sends as much junk as to fill the uncompressed size of the file it is sending.  

I.e. a gzip file which uncompressed is 44KB.  The server sends 44KB of the gzip file and then whatever bytes follow in the espfs image.  Modern browsers seem to ignore this extra data, but chromium 53 does not.  And it defeats the purpose of compression if we are padding the gzip file with junk so that the effective compression is 1:1.

I dug into it and found that the server just relies on the file size provided by espfs.
https://github.com/chmorgan/libesphttpd/blob/master/core/httpd-espfs.c#L202

I added some debug crumbs and espfs_stat also reports the incorrect (as if uncompressed) file size.
I.e. for the file /static/app-25efe0c865.min.css which is 44280 bytes uncompressed.  When compressed it should be less than half of that.
```
espfs_stat_t s = {0};
espfs_fstat(file, &s);
ESP_LOGD(TAG, "espfs_fstat size: %d flags: %x comp: %x", s.size, s.flags, s.compression);
```
`D (127935) httpdespfs: espfs_fstat size: 44280 flags: 3 comp: 0`

I'm guessing the bug is probably in the generating of the espfs image, which writes the file size to the header.  I'll keep digging...

phatpaul, 2022-10-12:
I found a fix that jkent added to a later version.  https://github.com/jkent/frogfs/blob/d2d70ade10abb6f82c9fe77766f069780b4aca3a/tools/mkfrogfs.py#L133

From commit: https://github.com/jkent/frogfs/commit/d2d70ade10abb6f82c9fe77766f069780b4aca3a#

I'll just cherry-pick this part to back-port to the V3 branch:
```
    if flags & ESPFS_FLAG_GZIP:
        file_len = data_len
```

jkent, 2023-08-08:
Resolved in v3 branch

closed 2023-08-08 by jkent