jkent/frogfs #43: MicroPython

jkent, 2023-08-22:
Test with micropython

Andrei-Pozolotin, 2023-12-06:
@jkent : Jeff: micropython is also very interested :)

https://github.com/micropython/micropython/pull/8381
https://github.com/micropython/micropython/pull/8381#issuecomment-1841993791

can you please share your test results? or perhaps there is a place for synergy?

jkent, 2023-12-06:
Very cool.  I don't have any test results/benchmarks at the moment, nor have I done anything with FrogFS + micropython.  I'm going to be working on finalizing the API in the "next" branch.  I learned a little when working on the ramfs code, and want to put this into practice before making things final.  I just now separated out the example code to their own repositories, which is a step forward in making this ready for an IDF component.

Discussion here or on the micropython issue is fine with me.  Feel free to ping me here or there.