jkent/frogfs #47: support python 3.8?

phatpaul, 2023-09-01:
preprocess.py:49 gives error:
AttributeError: 'str' object has no attribute 'removeprefix'
Looks like this was added in python 3.9, but I have 3.8.  

Since 3.8 is supported in IDF 5.1, I think it should be supported here too.

jkent, 2023-09-02:
fixed in 9a3dffb

closed 2023-09-02 by jkent

phatpaul, 2023-09-05:
Thanks.
It's a shame that the fix reduces the readability of the code.  The first line is self-documenting, the second, not so much.
```c
-  name, _ = os.path.splitext(file.removeprefix('transform-'))
+  name, _ = os.path.splitext(file[10:])
```
I'm curious if python has something like poly-fills (like in javascript) to prevent having to reduce the code quality to support older VM versions?

jkent, 2023-09-05:
Yeah, "monkey patching" is what they call it in Python.  Because there is the startswith(...) lines that guard it, I'm okay with the less than elegant solution.