jkent/frogfs #49: mkfrogfs.py fails if root dir has subdirectories starting with dot (next branch)

valkuc, 2023-10-19:
Hello.
The directory structure is like this:
`/.test_dot_dir/.test.html`

```
Result:
       - Stage 1
         - test_dot_dir/.test.html
Traceback (most recent call last):
  File "E:\Espressif\workspace\test_proj\managed_components\frogfs\tools\mkfrogfs.py", line 511, in <module>
    dirty |= run_preprocessors(entries)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Espressif\workspace\test_proj\managed_components\frogfs\tools\mkfrogfs.py", line 324, in run_preprocessors
    preprocess(entry)
  File "E:\Espressif\workspace\test_proj\managed_components\frogfs\tools\mkfrogfs.py", line 261, in preprocess
    with open(os.path.join(g_root_dir, path), 'rb') as f:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '\\test_dot_dir/.test.html'
```
It looks like the dot is removed from directory name in mkfrogfs.py

jkent, 2023-10-19:
Hmm, looks like changing `.lstrip('.').lstrip('/')` to `.lstrip('./')` on lines 112 and 136, that should solve that problem.  Could you try this and report back?

valkuc, 2023-10-19:
Looks like did not help. Nothing changed, same error. Actually, I'm thinking that it's a good idea to skip files (and directories) with name starting from "." and do not include them in FrogFs image. What do you think?

valkuc, 2023-10-19:
Btw, I tried to implement such exclusion by adding
```
    if reldir.startswith('.'):
            continue
```
at lines 113 and 139, but this not helped and I'm stuck. My Python knowledge is below zero :)

jkent, 2023-10-19:
fixed in 1a3d571bbe23db53e586a6211a9917324a5a3917

Thanks!

valkuc, 2023-10-19:
Thanks, now there is no errors and the directory with file is added to FrogFs image. Could you please tell me, is it possible to exclude any or specific directory from image?

jkent, 2023-10-19:
it *might* be broken at the moment, but you can try this:

You can do the following, which glob matches any files and folders with that name:

```yaml
filters:
  'path/to/.folder_name*':
    - discard
  ```

I'll create a bug, because you should be able to just do the following:

```yaml
filters:
  'path/to/.folder_name':
    - discard
 ```

But that is currently broken.

valkuc, 2023-10-19:
Looks like this doesn't work either.
In the log:
```
       - Stage 1
         - .test-data/state.json
           - compress deflate... skipped
```
but actually the file is included as I see it in binary.

jkent, 2023-10-19:
I'll look into this.  In the meantime, can you open a new issue with a snippet of your yaml?

valkuc, 2023-10-19:
Yep,will do