jkent/frogfs #51: support for mime types

jkent, 2023-10-19:
I've got a couple of ideas kicking around for managing mime types in FrogFS.

 1. A lookup table of uint16_t to mime type strings, with uint16_t field added to files.
 1. A file on the file system that maps file extensions to mime types

There are pros and cons to both approaches.  The former is fast, but requires modifying the binary format yet again. The later either requires memory caching of the mime types or slow string lookups, but it can be an optional feature.

valkuc, 2023-10-19:
p.1 looks interesting, I just thinking that uint8_t will be enough.

Btw, I'm still use old swiss-knife approach to get mime type by extension and... it works well and satisfies all needs.

jkent, 2023-10-19:
True, but C requires me to use 32-bit alignment, even in the LUT.  It seems wasteful.  Maybe the second idea would be for the best.

jkent, 2023-12-08:
I'm not going to implement this, because it is outside the scope of the filesystem.  I'll come up with a method to lookup the mimetype in cwhttpd when it is convenient.

closed 2023-12-08 by jkent