jkent/frogfs #52: how to use with VSCode + Platform.io?

hoeken, 2023-10-20:
This looks like a great library/project.  However, I'm using VSCode and Platform.io and not cmake.  Any idea on how I would go about integrating this?

jkent, 2023-10-20:
What device are you targeting?  If you're targeting ESP32, Platform.io should be using the IDF CMake build system behind the scenes.  You should be able to create a components directory within your project and checkout the FrogFS there.  Once I release the `next` branch, FrogFS will be put on the Espressif component registry, which may integrate with Platform.io easier.  Also, note that you must be using IDF version 5.0 or greater.  Let me know if you need further direction.

hoeken, 2023-10-20:
Yes, I'm using esp32.  I'll take a look at that, thanks.

hoeken, 2023-12-08:
Hey, thanks for reaching out. I ended up solving my issue with a gulp file that compresses all my html, css, js, and images into a single file, and turns that into an #include-able header.  https://github.com/hoeken/yarrboard-firmware/blob/main/gulpfile.mjs

Its working very nicely so far and solves my main issue of wanting a single .bin file for easy and fail-safe OTA updates.

closed 2023-12-08 by hoeken