jkent/frogfs #53: publishing at components.espressif.com

igrr, 2023-10-23:
Hi @jkent, I wonder if you could publish this component in the [IDF component registry](https://components.espressif.com/) to make it easier to add to IDF projects?

To do this, you only need to:
- add an idf_component.yml file ([example 1](https://github.com/joltwallet/esp_littlefs/blob/master/idf_component.yml), [example 2](https://github.com/igrr/pa1010d/blob/main/idf_component.yml), [docs](https://docs.espressif.com/projects/idf-component-manager/en/latest/reference/manifest_file.html))
- log into components.espressif.com with your Github account and create a token
- add a Github Actions workflow to publish the component for every new tag ([example](https://github.com/igrr/pa1010d/blob/main/.github/workflows/upload_component.yml))

(I have a [talk](https://www.youtube.com/watch?v=D86gQ4knUnc) & [slides](https://igrr.github.io/edc23/) on this topic, in case it helps.)

jkent, 2023-10-24:
Hey @igrr,  This is actually already planned when the `next` branch is merged into master.  I need to move the `examples` directory out of the tree, because clockwise has an unnecessary submodule.  Thank you for the friendly poke though!

igrr, 2023-10-24:
I have missed the `next` branch, thanks for the pointer!

> I need to move the examples directory out of the tree, because clockwise has an unnecessary submodule.

If you are worried about that being a problem for the component registry — it shouldn't be, as long as you check out the submodules before uploading the component. The component manager will simply upload all the available files, including those inside submodules, .git directories excluded.

jkent, 2023-12-08:
Published to the component registry as jkent/frogfs v4.0.1

Thanks for the help igrr!  ramfs and overlayfs are next.