jkent/frogfs #56: Compiler warning in vfs.c: free' discards 'const' qualifier

M-Bab, 2024-05-15:
At first thanks for your great project! Nice to see all its progress and continuously better integration as ESP component.

Quite simple issue here: We get a compile warning which can be easily avoided in https://github.com/jkent/frogfs/blob/bf6173f75974acb83f86118864d27064e72d39e1/src/vfs.c#L216

The target of name is dynamically allocated so not so constant after all. free(name) causes a warning because free wants (obviously) a non-constant input.

Several easy solutions:

1. `const char *name = frogfs_get_name(entry);` --> `char *name = frogfs_get_name(entry);`
2. `free(name)` --> `free((char*) name)`
3. `free(name)` --> `free((void*) name)`

I can create a PR for this if it helps but I am not sure which is your favorite solution.

jkent, 2024-06-30:
Hi M-Bab,

Thanks for bringing this to my attention.  I think I actually have this as an uncommitted change that fell through the cracks.  Since `name` should be immutable, the correct way to handle this is `free((char *) name);` or `free((void *) name);`.  I'd go with the later, because that matches the prototype of the free function.  Feel free to make a PR.

closed 2024-08-04 by jkent

M-Bab, 2024-08-05:
Thanks a lot, sorry I missed to make a PR for this.

commit d4f02856 (2024-08-04): Fix free of const pointer warning, closes #56