jkent/frogfs #57: IDF build error after reconfigure

pchevali, 2024-06-04:
The python cache is generated in the frogfs directory thus triggering an IDF error after a reconfigure when using the component.

ERROR: Some components (jkent/frogfs) in the "managed_components" directory
  were modified on the disk since the last run of the CMake.  Content of this
  directory is managed automatically.

To reproduce:
1) add frogfs as project dependancy (idf.py add-dependency jkent/frogfs)
2) Build a project with frogfs filesystem : idf.py build
3) Clean the project : idf.py clean
4) Rebuild the project : idf.py build

This triggers the error mentioned above.

jkent, 2024-06-30:
Hi pchevali,

Thanks for bringing this to my attention.  I'll have to triage this for later.  In the meantime, if you have found a resolution to this, I'd love to hear it.

pchevali, 2024-06-30:
Hi,
You can find my pull request here #59 
I added the -B command line option to the python  command, this avoid to generate the pycache files. Although we can use a Python environment variable to fix the issue with the IDF, using a command line option can give a better out of the box experience.

jkent, 2024-06-30:
Thank you, merged.

closed 2024-06-30 by jkent