jkent/frogfs #65: Options out-of-phase or not supported

BrucePerens, 2025-02-17:
Hi,

Thanks for FrogFS.

The example configuration file doesn't work any longer, at least when I build it (on Debian 12.9). If I give it:

filter:
  '*':
    - compress zlib
      level: 9

The yaml parser complains that mapping values aren't allowed there. If I take away the colon after "level", it works. Or at least doesn't complain.

The html-minifier doesn't work, it seems due to depending on a Python html-minifier package that isn't current.

    Thanks

    Bruce

BrucePerens, 2025-02-17:
Also, it's important to note in the docs that collect directories should be entered with a trailing slash, or their name will become part of the pathname in the compiled filesystem.

jkent, 2025-02-18:
> The example configuration file doesn't work any longer, at least when I build it (on Debian 12.9). If I give it:
--- snipped ---
> The yaml parser complains that mapping values aren't allowed there. If I take away the colon after "level", it works. Or at least doesn't complain.

Try indenting level 2 more spaces.  YAML is very particular about indentation.  I believe without the extra two spaces, it doesn't create a dictionary.

> Also, it's important to note in the docs that collect directories should be entered with a trailing slash, or their name will become part of the pathname in the compiled filesystem.

Good catch.  Do you have time to make a PR on this?

BrucePerens, 2025-02-19:
Done. https://github.com/jkent/frogfs/issues/66

jkent, 2025-07-01:
I confirmed that this was a YAML syntax issue.

The correct syntax is:

```yaml
filter:
  '*':
    - compress zlib:
        level: 9
```

Notice the colon, and the indentation.

Sorry I did not get around to this sooner.

closed 2025-07-01 by jkent